CMU-SAFARI/ramulator2 #7: Details on how to intergrate ramulator2.0 to gem5

brightspiritme, 2023-09-09:
In the README document, the fourth step in the instructions for integrating emulator2.0 into gem5 mentions "Put the Ramulator2 wrapper code to gem5/src/mem/", but the only relevant code found in the source code is "src/frontend/impl/external_wrapper/gim5_frontend. cpp", and this code seems incomplete.
And the fifth step also mentioned the need to register "Ramulator2. py" as the SimObject of Gem5, but there is no reference to Ramulator2. py in the source code.
May I ask if anyone can provide references for ramulator2. cc, ramulator2. hh, and Ramulator.py? I would greatly appreciate it!

RichardLuo79, 2023-09-19:
Sorry for the inconvenience. We missed some files when preparing the public version. 

We have added the missing files back at [gem5_wrappers](https://github.com/CMU-SAFARI/ramulator2/tree/main/resources/gem5_wrappers)

Note that the gem5 frontend (`src/frontend/impl/external_wrapper/gim5_frontend. cpp`) is complete. It is a "dummy" frontend in Ramulator 2 that serves the purpose of receiving memory requests from gem5 and forward them to the memory system. Hope this helps!

closed 2023-09-19 by RichardLuo79

brightspiritme, 2023-11-05:
Thank you so much!