CMU-SAFARI/ramulator2 #12: fatal error: ranges: No such file or directory

Nakul1048, 2023-09-25:
Hi,

When I ran the 'make -j' command, it reports error "fatal error: ranges: No such file or directory
    9 | #include <ranges>
      |          ^~~~~~~~
compilation terminated."

Do you have any idea about it?
Thank you.

RichardLuo79, 2023-10-18:
Hi,

I could not reproduce this error on my side. Could you provide more information on the build environment (e.g., compiler toolchain version, cmake version, make version)?

prapti02, 2024-02-06:
Hi, 

I am getting the same error. When I ran the 'make -j' command, it reports error "fatal error: ranges: No such file or directory
9 | #include
| ^~~~~~~~
compilation terminated."

Thanks,

praisechan, 2024-10-19:
Hi, @Nakul1048 

could you please let me know how you solved this problem?

yingj4, 2024-12-17:
`ranges` is not included in `gcc-9`. You need to use `gcc-10` or above.