CMU-SAFARI/ramulator2 #13: Stuck at 93%

Ranyang-Zhou, 2023-09-26:
Hi everyone,
I followed the instruction and typed 'make -j', the system stucked at 93%

"[ 93%] Building CXX object _deps/yaml-cpp-build/CMakeFiles/yaml-cpp.dir/src/tag.cpp.o"

Then the system dumped and nothing was installed.

Does anyone can give me an idea? Thanks a lot.

ankenman, 2023-09-28:
It seems like you're stuck building the YAML library. Can you build it separately? Which CMake and compiler versions are you using?

RichardLuo79, 2023-10-18:
Hi Ranyang,

What compiler toolchain are you using?
You could try removing everything under /ext and reconfigure & build the project.

zifeng-yang, 2023-11-23:
Hi Ranyang,

I had a similar problem to yours, and I used “make -j2” then make ramulator successfully.
Try to add an even number after make j, if your core number more than one.

RichardLuo79, 2024-01-27:
We were able to reproduce this issue on some computers.

It looks like `make -j` could cause very high CPU utilization during compilation, causing some systems to freeze. To avoid this, you can compile with less number of concurrent jobs (less than the number of cores of your CPU).

closed 2024-01-27 by RichardLuo79