CMU-SAFARI/ramulator2 #14: LPDDR5 bandwidth not matching expectations

ankenman, 2023-09-27:
I am having two issues when using LPDDR5 with an external simulator:
- Sequential and Random streams give the same bandwidth (expect random to be ~2/3; DDR4 works correctly)
- Bandwidth is much greater than what I expect. How big are the data chunks for each callback from
`ramulator2_frontend->receive_external_requests()`?

RichardLuo79, 2023-10-18:
Hi,

Thanks for spotting this issue!
- May I know what configurations you are using for the LPDDR5 experiments?
- `receive_external_requests()` has no fixed transaction size on its own. The memory system (i.e., controller and DRAM) decides the number of bytes transferred per transaction (e.g., `m_channel_width` * burst_size).

ankenman, 2023-10-18:
My YAML file is similar to the example provided:

```
  Frontend:
    impl: GEM5
  
    Translation:
      impl: RandomTranslation
      max_addr: 2147483648
                
  
  MemorySystem:
    impl: GenericDRAM
    clock_ratio: 4
  
    DRAM:
      impl: LPDDR5
      org:
        preset: LPDDR5_2Gb_x16
        channel: 1
        rank: 1
      timing:
        preset: LPDDR5_6400
  
    Controller:
      impl: Generic
      Scheduler:
        impl: FRFCFS
      RefreshManager:
        impl: AllBank
      plugins:
  
    AddrMapper:
      impl: RoBaRaCoCh
```

How would I go about setting `m_channel_width` and burst_size?

RichardLuo79, 2024-02-11:
Hi, 

We recently pushed some bug fixes involving state updates and timing parameters in LPDDR5 (#27 and #32). Could you please let us know if these fixes solve your issue? Thanks!

closed 2026-03-25 by RichardLuo79