CMU-SAFARI/ramulator2 #15: m_internal_prefetch_size for LPDDR5

tongCIX, 2023-10-04:
Could you please explain the m_internal_prefetch_size here?
DDR5 has 16, however LPDDR has 8?

RichardLuo79, 2024-01-27:
Hi,

`m_internal_prefetch_size` refer to how much data are fetched for burst RD/WR operation of the DRAM chip (in the units of the channel width). This is used to calculate the amount of bytes that are transferred for one transaction:
```c++
int tx_bytes = m_dram->m_internal_prefetch_size * m_dram->m_channel_width / 8;
```
For example, DDR4 has internal_prefetch_size of 8, then a rank of DRAM chips will fetch 8 * 64 bits of data from the array for an RD/WR. These data will be sent in 8 bursts.

DRAM standards introduce higher internal prefetch size to bridge the gap between the slow memory core array and fast I/O. More information can be found here:
https://www.synopsys.com/blogs/chip-design/ddr-generations-memory-density-speed.html

https://en.wikipedia.org/wiki/Synchronous_dynamic_random-access_memory#DDR_SDRAM_prefetch_architecture

RichardLuo79, 2026-03-25:
Hi, we have Ramulator 2.1 on the [v2.1 branch](https://github.com/CMU-SAFARI/ramulator2/tree/v2.1) that provided comprehensive support for LPDDR5 WCK2CK sync (with expiry tracking) and tAAD deadline aware scheduling of ACT1 and ACT2. We also fixed the LPDDR5 internal prefetch size to be 16. Please check it out and let us know if you find any other problem!