CMU-SAFARI/ramulator2 #16: DDR5 RFM is not specified

jyk2498, 2023-10-06:
Hi everyone

I modified the 'example_config.yaml' file to run DDR5, but when I try to run it, I encounter the following error message:

`terminate called after throwing an instance of 'Ramulator::ConfigurationError'
  what():  ParamGroup "RFM" is not specified for implementation "DDR5".
Aborted (core dumped)`

How can I resolve this issue?

Thanks.

RichardLuo79, 2023-10-18:
Hi,

Could you share the YAML configuration file here?

ankenman, 2023-10-18:
I can reproduce this issue with this config:

```
Frontend:
  impl: GEM5

  Translation:
    impl: RandomTranslation
    max_addr: 2147483648
              

MemorySystem:
  impl: GenericDRAM
  clock_ratio: 4

  DRAM:
    impl: DDR5
    org:
      preset: DDR5_8Gb_x4
      channel: 1
      rank: 1
    timing:
      preset: DDR5_3200AN

  Controller:
    impl: Generic
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank
    plugins:

  AddrMapper:
    impl: RoBaRaCoCh
```

jyk2498, 2023-10-21:
below is the yaml file that I ran:
```
Frontend:
  impl: SimpleO3
  clock_ratio: 8
  num_expected_insts: 300000
  traces: 
    - example_inst.trace

  Translation:
    impl: RandomTranslation
    max_addr: 2147483648
              

MemorySystem:
  impl: GenericDRAM
  clock_ratio: 3

  DRAM:
    impl: DDR5
    org:
      preset: DDR5_16Gb_x8
      channel: 1
      rank: 2
    timing:
      preset: DDR5_3200AN

  Controller:
    impl: Generic
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank
    plugins:

  AddrMapper:
    impl: RoBaRaCoCh
```

jarrett-m, 2023-11-15:
in the config you need to specify what RMF you want to use, you need the param but if you dont add anything after it just defaults

`Frontend:
  impl: SimpleO3
  clock_ratio: 8
  num_expected_insts: 300000
  traces:
    - example_inst.trace

  Translation:
    impl: RandomTranslation
    max_addr: 2147483648

MemorySystem:
  impl: GenericDRAM
  clock_ratio: 3

  DRAM:
    impl: DDR5
    org:
      preset: DDR5_8Gb_x4
      channel: 1
      rank: 2
    timing:
      preset: DDR5_3200AN
    RFM:
      #this will choose a default RFM in ddr5.cpp, but you need the RFM: part

  Controller:
    impl: Generic
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank
    plugins:

  AddrMapper:
    impl: RoBaRaCoCh
`

closed 2024-01-27 by RichardLuo79