CMU-SAFARI/ramulator2 #17: Error when integrating to gem5

Mi5sssss, 2023-10-06:
Hi,
I followed the steps to integrate ramulator2 to gem5. I got following errors when I try to build the gem5:

`src/mem/ramulator2.cc:5:10: fatal error: debug/Ramulator2.hh: No such file or directory
    5 | #include "debug/Ramulator2.hh"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [build/X86/mem/ramulator2.do] Error 1
scons: building terminated because of errors.`

RichardLuo79, 2023-10-18:
Hi,

What compiler toolchain are you using?

cecetaca, 2023-12-13:
Hi, 

I'm having the same problem. Looks like it's expecting build/<ARCH>/debug to contain Ramulator2.hh?
Changing the arch or build type [debug, opt, fast] doesn't change anything either.
My compiler is gcc version 10.5.0

Help would be appreciated!

sangjae4309, 2024-01-10:
You're missing the debug flag-Ramulator2-so that "debug/Ramulator2.hh" is not generated.

Add  following code to `src/mem/SConscript` 

````
if env['HAVE_RAMULATOR2']:
    SimObject("Ramulator2.py", sim_objects=['Ramulator2'])
    Source("ramulator2.cc")
    DebugFlag("Ramulator2")
```

cecetaca, 2024-01-10:
Thank you!
I see that wasn't reflected on the README.

Still leads to this error on compile time, though:

`build/ARM/mem/ramulator2.cc: In constructor ‘gem5::memory::Ramulator2::MemorySystemPort::MemorySystemPort(const string&, gem5::memory::Ramulator2&)’:
build/ARM/mem/ramulator2.cc:314:50: error: no matching function for call to ‘gem5::ResponsePort::ResponsePort(const string&)’
  314 |     : ResponsePort(_name), ramulator2(_ramulator2)`

sangjae4309, 2024-01-13:
Your bug message indicates that there is no constructor of _ResponsePort_, which requires only single string argument. 
Check the constructor at `gem5/src/mem/port.hh`.
I think the version gem5 you're using is obsolete. please update the gem5 to latest one.
For your information, i'm currently using gem5-Aug.30.2023.

cecetaca, 2024-01-16:
I confirm that after the changes in the pull request by @sangjae4309 everything is solved.
I suggest modifying the README to clarify directory structure, etc. as pointed out above, which was the main problem.

RichardLuo79, 2024-01-27:
Thank you. We have updated the README

closed 2024-01-27 by RichardLuo79

The-CliffWu, 2024-12-08:
Hi, Can I ask the result of combining ramulator2 with gem5?

For this case, I used DDR4_2400M_x8, which operates at a DRAM frequency of 1200 MHz. The ramulator.dram_cycles suggests that the DRAM component accounts for approximately 50% of the total simulation time for the system. However, when I applied the formula, I found that the DRAM component consistently accounts for 50% of the total simulation time, regardless of the architecture or neural network model size, whether big or small. This seems odd and makes me suspect that there might be an issue with the formula or conclusion I previously mentioned.