CMU-SAFARI/ramulator2 #21: Verifying Ramulator 2.0's Memory Controller and DRAM Model

alinezhad2018, 2023-10-27:
Hi,

I am trying to verify the ramulator memory controller and DRAM model as explained in the README.
I am using ncsim (ncverilog) as a simulator -  ddr4_verilog_models/protected_ncverilog-

I modified these files "tb.sv" and also "subtest.vh" in the ddr4_verilog_models, to support the ramulator configuration. These changes were necessary to align the simulation with Ramulator's configuration.

When I use the default period setting (where default_period(nominal_ts) = TS_1250, corresponding to a data rate of 1600) in the "subtest.vh" file, I can successfully run the Ramulator traces generated with DDR4-2400. However, when I adjust the period and set it to TS_833, I encounter a VIOLATION ERROR.

For example, one of the errors I encounter is:

_tb.golden_model.always_diff_ck.if_diff_ck:VIOLATION: cmdRD BG:0 B:0 A:2 (BL:4 WL:10 RL:12) @2820.0 ns Required:
tWTR_L (CWL + BL/2 + tWTRc_L) - 4 clocks._
Or:
_tb.golden_model.VerifyMR:ERROR:SPEC_VIOLATION tWR/tRTP tWR spec:19 loaded:14 tRTP spec:5831 loaded:5831 @2137.5 ns_

I'm wondering if you could provide an example of a trace file that you generated with Ramulator and ran on the DRAM model without encountering any violations.
Furthermore, I would appreciate more detailed guidance on how to modify "tb.sv" and "subtest.vh" within the ddr4_verilog_models to match the Ramulator DDR configuration and run simulations without errors.

RichardLuo79, 2024-01-27:
Hi Saeideh,

We have updated the repo to include the already configured verilog model (in verilog_verification/sources) and some scripts to convert the Ramulator 2 traces to testbenches and check the simulation output for errors. We have also updated the README to include mode detailed instructions.

Please let us know if you encounter any further issue, thanks!

alinezhad2018, 2024-02-02:
Thank you!
I've examined the updated source folder, and it executed smoothly without encountering any VIOLATION or ERROR (for DDR4_2400 ).
I'd like to extend this information to others who might be interested in replicating the same experiments. It's crucial to ensure that the count of instructions specified in "verification-config.yaml" aligns with the number of instructions generated by "tracegen.py". Failing to maintain this correspondence might lead to potential VIOLATION issues.