CMU-SAFARI/ramulator2 #23: Issues related to the output of simulation results

brightspiritme, 2023-11-05:
Hi!

Compared to the ramulator v1, the stat report output at the terminal lacks many parameter indicators, such as bandwidth, read/write latency, etc. Is it possible to add more simulation information and output the report to a user defiend file. I will be approciate it!

RichardLuo79, 2024-01-27:
Hi,

We are working on adding more & better statistics output to Ramulator 2. 

Meanwhile, if you need some statistics urgently, you can easily add them: Take a look at the examples in [example_impl.cpp](https://github.com/CMU-SAFARI/ramulator2/blob/main/src/example/impl/example_impl.cpp)

RichardLuo79, 2026-03-25:
Hi, we have Ramulator 2.1 on the [v2.1 branch](https://github.com/CMU-SAFARI/ramulator2/tree/v2.1) that significantly improved configuration and stats organization. Please check it out!

closed 2026-03-25 by RichardLuo79