CMU-SAFARI/ramulator2 #25: comparing ramulator1 and ramualtor2 with the same configs

alinezhad2018, 2023-11-13:
Hi,

I am trying to run the same trace using ramualtor1 and ramualtor2, but even by using the same config files as you provided in the "perf_comparison" folder,
https://github.com/CMU-SAFARI/ramulator2/tree/main/perf_comparison/configs

The final results (such as memory_system_cycle) are not exactly the same for both of the simulations; however, they are very close.
I expected to see exactly the same final results for both ramualtor1 and ramualtor2, as they implement the same memory with almost the same configuration. I wonder if there is any change in the implementation that may lead to different results. 

I'm curious to know if you've observed similar outcomes in your experiments. I've noticed that the reference paper only provides a comparison of "simulation time" without any reference to the final results achieved by different simulators.

Thanks!
Saeideh

RichardLuo79, 2024-01-27:
Hi Saeideh,

Yes, we also observe a slight difference in the number of cycles in our experiments. The reason for this is that the memory controller implementation of Ramulator 2 used in the experiment contains improvements and bug fixes over Ramulator 1's memory controller. 

For example, in Ramulator 1, the scheduler can close the row that a memory request just opened, *before* issuing the RD/WR command to serve the request, causing "useless" activations. Ramulator 2's generic memory controller implementation [fixes this issue](https://github.com/CMU-SAFARI/ramulator2/blob/12cd50be97e8a0b7be5185f865edbd6c20c75a3e/src/dram_controller/impl/generic_dram_controller.cpp#L225).

Hope this helps!

alinezhad2018, 2024-01-29:
Thank you for your reply.
Does this imply that the results from Ramulator2 are more precise than those from Ramulator1?
It's possible that the simulation time and the number of cycles are influenced by these bug fixes.

 Additionally, I noticed a difference in **DRAM cycles** between Ramulator1 and Ramulator2 based on the type of traffic (random or sequential). 
For random traffic, Ramulator1 shows more DRAM cycles than Ramulator2, whereas for sequential traffic, Ramulator2 exhibits more DRAM cycles compared to Ramulator1 (DDR3_1600). 
 Have you also observed this pattern, and do you have any insights into the reason behind it, such as potential differences in implementation?
  
[image]

RichardLuo79, 2024-01-29:
Hi Saeideh,

> Does this imply that the results from Ramulator2 are more precise than those from Ramulator1?
With respect to the DRAM controller, it is more "precise" because a good DRAM controller should not send "useless" activations to DRAM banks.

> For random traffic, Ramulator1 shows more DRAM cycles than Ramulator2, whereas for sequential traffic, Ramulator2 exhibits more DRAM cycles compared to Ramulator1 (DDR3_1600).
We changed some timing values in the speed presets in Ramulator 2 to be more aligned with the JEDEC standards, compared to Ramulator 1. This could be causing the slight variations in the number of DRAM cycles.

closed 2024-02-01 by alinezhad2018