CMU-SAFARI/ramulator2 #37: Writing Your Own Wrapper of Ramulator 2.0 for my own Simulator

prapti02, 2024-02-10:
I am trying to do step 4 under 'Writing Your Own Wrapper of Ramulator 2.0 for my own Simulator' in readme file, to send the memory requests from my simulator to Ramulator 2.0. 
Inside frontend.h file, ("virtual bool receive_external_requests(int req_type_id, Addr_t addr, int source_id, std::function<void(Request&)> callback) { return false; }"), it returns false and hence the memory requests are not sent properly to the ramulator. 
Do I have to modify any files of the ramulator code base? Or do I have to make 2 separate files of MyWrapper.h and MyWrappr.cpp?

ankenman, 2024-02-10:
I have an example SystemC frontend on my fork of this repo. It may be useful for your development, and feel free to reach out to me for help. It's in the Readme:

https://github.com/ankenman/ramulator2

RichardLuo79, 2024-02-11:
Hi,

When you implement your own frontend that inherits from the frontend interface, your own implementation of this virtual function will override the default implementation in the interface class.

prapti02, 2024-02-15:
> I have an example SystemC frontend on my fork of this repo. It may be useful for your development, and feel free to reach out to me for help. It's in the Readme:
> 
> https://github.com/ankenman/ramulator2

I appreciate your help. I am trying the solution you gave to me. So, here is what I do:
1. I installed systemC.2.3.4. I ran an example systemC code to test the installation and it works.
2. Then inside the ramulator codebase, I copy your example wrapper ramulator cpp file. 
3. Then I am trying to modify the existing Cmakelist.txt file and here is where I am getting linking errors.

Is this the right way to run your example? Can you please provide me the steps to run your example ramulator wrapper file?

Thank,
Prapti