CMU-SAFARI/ramulator2 #40: Gem5 ARM full-system error

sudhanshu2, 2024-03-24:
I replaced the [memory config line of code](https://github.com/gem5/gem5/blob/e8bc4fc137a5a7e82b601432271a027b652ae69b/configs/example/arm/starter_fs.py#L111) in the starter full-system simulation for ARM with the code specified in the readme but get the following error:
```
src/mem/physical.cc:108: fatal: fatal condition addrMap.insert(m->getAddrRange(), m) == addrMap.end() occurred: Memory address range for system.realview.bootmem is overlapping
```
I was wondering if y'all knew why Gem5 is throwing this error with Ramulator 2 and/or have the Ramulator2 config and the Gem5 options that y'all used to verify that your code works?

Thanks :)

lnghrdntcr, 2024-06-15:
I second this, I'm encountering the same error when I'm trying to run a configuration that has multiple gem5 memory channels. Did you manage to solve this?

atrah22, 2024-08-12:
Same here. 
Could anyone manage to solve this?
There is no way to set the AddrRange in the Ramulator2() class. This is an issue due to to AddrRange which should start from 0x80000000 address for DDR as per the Express_GEM5_Base class in gem5/src/dev/arm/Realview.py.
If this is not resolved or shown in README file, its hard to compare ramulator with gem5 O3 ARM CPU

sangjae4309, 2024-09-19:
To set the AddrRange in the Ramulator2() class, We need to wrap the ramulator2 around AbstractMemorySystem which is conventional gem5's main memory source code.

Here is [my code](https://github.com/sangjae4309/gem5-ramulator2/blob/main/wrapper/python/ramulator_2.py) for how to set the addr range (line 67-82)

sangjae4309, 2024-09-19:
Any way, I think the error is originated from using wrong version of kernel or bootloader (not form setting AddrRange).

Look at [my script](https://github.com/sangjae4309/gem5-ramulator2/blob/main/script/base-system-fs-arm.py) (line77-79) that ARM full system simulation works properly.