CMU-SAFARI/ramulator2 #47: Question about clock ratio

zzczzc20, 2024-06-14:
Dear all,
In the standalone mode, there is a parameter called clock ratio. I want to know whether it uses the clock of memory or memory controller.

For example, if the CPU frequency is 2.8GHz, and I use DDR%-5600. If it is the clock of the memory it should be 1:1. If it is the clock of memory controller, the ratio should be 2:1, let us say the clock frequency of memory controller is 1.4GHz.

idealgao, 2024-09-05:
Same question. In SoC design field, these two ratio should be(guess), further testing...

Frontend: clock_ratio: -> sys_clk(cpu/bus_clk) : dfi_clk(ddrc_clk)
MemorySystem: clock_ration -> dfi_clk(ddrc_clk) : DRAM_CK(DDR4_3200=1600MHz)

RichardLuo79, 2026-03-26:
Hi,

In Ramulator, the memory-side clock is tCK (except for HBM3 where we have 0.5tCK per tick to handle 1.5tCK commands). So your understanding is correct: For DDR4 3200MT/s, the frequency is 1600MHz