CMU-SAFARI/ramulator2 #58: LPDDR5 bug with example_config.yaml

LIUQyou, 2024-09-03:
I build the ramulator2 in the ubuntu22.04 system.
I try to config the LPDDR5 using the following config file:
```yaml
Frontend:
  impl: SimpleO3
  clock_ratio: 8
  num_expected_insts: 500000
  traces: 
    - example_inst.trace

  Translation:
    impl: RandomTranslation
    max_addr: 2147483648

MemorySystem:
  impl: GenericDRAM
  clock_ratio: 3

  DRAM:
    impl: LPDDR5
    org:
      preset: LPDDR5_4Gb_x16
    timing:
      preset: LPDDR5_6400

  Controller:
    impl: Generic
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank
    RowPolicy:
      impl: OpenRowPolicy
      cap: 4
    plugins:

  AddrMapper:
    impl: RoBaRaCoCh

```
I try to run the example by using the following command: 
```shell
./ramulator2 -f example_config.yaml
```
The following is the result:
```
[Ramulator::Base] [debug] Registering interface MemorySystem...
[Ramulator::Base] [debug] Registering interface Frontend...
[Ramulator::Base] [debug] Registering interface TestIfce...
[Ramulator::Base] [debug] Registering implementation TestImpl2 to interface TestIfce...
[Ramulator::Base] [debug] Registering implementation TestImpl to interface TestIfce...
[Ramulator::Base] [debug] Registering implementation LoadStoreTrace to interface Frontend...
[Ramulator::Base] [debug] Registering implementation ReadWriteTrace to interface Frontend...
[Ramulator::Base] [debug] Registering interface Translation...
[Ramulator::Base] [debug] Registering implementation SimpleO3 to interface Frontend...
[Ramulator::Base] [debug] Registering interface DRAM...
[Ramulator::Base] [debug] Registering interface BHMemorySystem...
[Ramulator::Base] [debug] Registering implementation BHO3 to interface Frontend...
[Ramulator::Base] [debug] Registering implementation GEM5 to interface Frontend...
[Ramulator::Base] [debug] Registering implementation NoTranslation to interface Translation...
[Ramulator::Base] [debug] Registering implementation RandomTranslation to interface Translation...
[Ramulator::Base] [debug] Registering interface BHScheduler...
[Ramulator::Base] [debug] Registering interface Scheduler...
[Ramulator::Base] [debug] Registering interface ControllerPlugin...
[Ramulator::Base] [debug] Registering interface RefreshManager...
[Ramulator::Base] [debug] Registering interface RowPolicy...
[Ramulator::Base] [debug] Registering interface Controller...
[Ramulator::Base] [debug] Registering interface BHDRAMController...
[Ramulator::Base] [debug] Registering interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation BHDRAMSystem to interface MemorySystem...
[Ramulator::Base] [debug] Registering implementation DummyMemorySystem to interface MemorySystem...
[Ramulator::Base] [debug] Registering implementation GenericDRAM to interface MemorySystem...
[Ramulator::Base] [debug] Registering implementation ChRaBaRoCo to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation RoBaRaCoCh to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation MOP4CLXOR to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation ChRaBaRoCo_with_rit to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation RoBaRaCoCh_with_rit to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation MOP4CLXOR_with_rit to interface AddrMapper...
[Ramulator::Base] [debug] Registering implementation DDR3 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR4 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR4-VRR to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR4-RVRR to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR5 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR5-VRR to interface DRAM...
[Ramulator::Base] [debug] Registering implementation DDR5-RVRR to interface DRAM...
[Ramulator::Base] [debug] Registering implementation LPDDR5 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation GDDR6 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation HBM to interface DRAM...
[Ramulator::Base] [debug] Registering implementation HBM2 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation HBM3 to interface DRAM...
[Ramulator::Base] [debug] Registering implementation BHDRAMController to interface BHDRAMController...
[Ramulator::Base] [debug] Registering implementation DummyController to interface Controller...
[Ramulator::Base] [debug] Registering implementation Generic to interface Controller...
[Ramulator::Base] [debug] Registering implementation PRACDRAMController to interface BHDRAMController...
[Ramulator::Base] [debug] Registering implementation BHScheduler to interface BHScheduler...
[Ramulator::Base] [debug] Registering implementation BlockingScheduler to interface BHScheduler...
[Ramulator::Base] [debug] Registering implementation FRFCFS to interface Scheduler...
[Ramulator::Base] [debug] Registering implementation BLISS to interface BHScheduler...
[Ramulator::Base] [debug] Registering implementation PRACScheduler to interface BHScheduler...
[Ramulator::Base] [debug] Registering implementation AllBank to interface RefreshManager...
[Ramulator::Base] [debug] Registering implementation OpenRowPolicy to interface RowPolicy...
[Ramulator::Base] [debug] Registering implementation ClosedRowPolicy to interface RowPolicy...
[Ramulator::Base] [debug] Registering implementation TraceRecorder to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation CommandCounter to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation PARA to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation Graphene to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation OracleRH to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation TWiCe-Ideal to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation Hydra to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation RRS to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation AQUA to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation RFMManager to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation BlockHammer to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation BLISS to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering implementation PRAC to interface ControllerPlugin...
[Ramulator::Base] [debug] Registering interface ExampleInterface...
[Ramulator::Base] [debug] Creating an implementation of interface Frontend...
[Ramulator::Base] [debug] Creating implementation RandomTranslation of interface Translation...
[Ramulator::Base] [debug] Creating an implementation of interface MemorySystem...
[Ramulator::Base] [debug] Creating implementation LPDDR5 of interface DRAM...
[Ramulator::Base] [debug] Creating implementation RoBaRaCoCh of interface AddrMapper...
[Ramulator::Base] [debug] Creating implementation Generic of interface Controller...
[Ramulator::Base] [debug] Creating implementation FRFCFS of interface Scheduler...
[Ramulator::Base] [debug] Creating implementation AllBank of interface RefreshManager...
[Ramulator::Base] [debug] Creating implementation OpenRowPolicy of interface RowPolicy...
[Ramulator::SimpleO3] [info] Processor Heartbeat 10000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 20000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 30000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 40000000 cycles.
```
The ramulator is built with debug mode. It will never end

anaithani-imachines, 2025-07-31:
What's the LPDDR5 config you used? Did you change something? I ran with the same config and command, but I get the error

[Ramulator::SimpleO3] [info] Processor Heartbeat 10000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 20000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 30000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 40000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 50000000 cycles.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to send refresh!
Aborted (core dumped)

naithanianshul, 2025-07-31:
What's the LPDDR5 config you used? Did you change something? I ran with the same config and command, but I get the error

[Ramulator::SimpleO3] [info] Processor Heartbeat 10000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 20000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 30000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 40000000 cycles.
[Ramulator::SimpleO3] [info] Processor Heartbeat 50000000 cycles.
terminate called after throwing an instance of 'std::runtime_error'
what(): Failed to send refresh!
Aborted (core dumped)