CMU-SAFARI/ramulator2 #59: ReadWriteTrace is not working properly, the output is not correct

ssakash6611, 2024-09-09:
yaml file:
Frontend:
  impl: ReadWriteTrace
  path: /home/ssa/ramulator2/readwriteaccessFirstTry3.trace
  clock_ratio: 3

  Translation:
    impl: RandomTranslation
    max_addr: 1140736775975936
              

MemorySystem:
  impl: GenericDRAM
  clock_ratio: 3

  DRAM:
    impl: DDR4
    org:
      preset: DDR4_8Gb_x8
      channel: 1
      rank: 2
    timing:
      preset: DDR4_2400R 

  Controller:
    impl: Generic
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank
    RowPolicy:
      impl: ClosedRowPolicy
      cap: 4
    plugins:

  AddrMapper:
    impl: RoBaRaCoCh
    
    
Input file "readwriteaccessFirstTry3.trace" contains (the decimal are converted virtual addresses from Intel Pintool):
W 140736775976056,140736775976056
W 140736775976048,140736775976048
W 140736775976040,140736775976040
W 140736775976032,140736775976032
W 140736775976024,140736775976024
W 140736775975928,140736775975928
W 140736775975936,140736775975936
R 140189629292038,140189629292038
W 140189629292038,140189629292038
W 140189629291184,140189629291184
W 140189629287008,140189629287008
R 140189629288024,140189629288024
W 140189629291200,140189629291200
W 140189629291360,140189629291360
R 140189629288040,140189629288040
W 140189629291280,140189629291280
R 140189629288056,140189629288056
W 140189629291912,140189629291912
R 140189629288072,140189629288072
W 140189629291288,140189629291288    

Command:  ./ramulator2 -f ./example_config.yaml

Output:
[Ramulator::ReadWriteTrace] [info] Loading trace file /home/ssa/ramulator2/readwriteaccessFirstTry3.trace ...
[Ramulator::ReadWriteTrace] [info] Loaded 20 lines.
Frontend:
  impl: ReadWriteTrace

MemorySystem:
  impl: GenericDRAM
  total_num_other_requests: 0
  total_num_write_requests: 1
  total_num_read_requests: 0
  memory_system_cycles: 0
  DRAM:
    impl: DDR4
  AddrMapper:
    impl: RoBaRaCoCh

  Controller:
    impl: Generic
    id: Channel 0
    avg_read_latency_0: .nan
    read_queue_len_avg_0: .nan
    write_queue_len_0: 0
    queue_len_0: 0
    num_other_reqs_0: 0
    num_write_reqs_0: 1
    read_latency_0: 0
    priority_queue_len_avg_0: .nan
    row_hits_0: 0
    priority_queue_len_0: 0
    row_misses_0: 0
    row_conflicts_0: 0
    read_row_misses_0: 0
    queue_len_avg_0: .nan
    read_row_conflicts_core_0: 0
    read_row_hits_0: 0
    write_queue_len_avg_0: .nan
    read_row_conflicts_0: 0
    write_row_misses_0: 0
    write_row_conflicts_0: 0
    read_queue_len_0: 0
    write_row_hits_0: 0
    read_row_hits_core_0: 0
    read_row_misses_core_0: 0
    num_read_reqs_0: 0
    Scheduler:
      impl: FRFCFS
    RefreshManager:
      impl: AllBank

    RowPolicy:
      impl: ClosedRowPolicy
      num_close_reqs: 0

Other files are not changed.

ssakash6611, 2024-09-10:
Greetings! I hope you are well. I am Sadman Sakib Akash, currently working as a GRA at University of Central Florida.
Regarding the problem above, I also used virtual addresses directly. Whatever the input the output is always same. what am I missing? Do I need to change anything in the code to process the input for calculating memory timing and power.

luzhixing12345, 2025-07-13:
Hi @ssakash6611 👋

Thanks for raising this issue — I recently ran into the same problem when using ReadWriteTrace, and after digging into the source code, I believe I’ve located the root cause. I’d like to ask whether you've already solved it, and also share some technical observations and suggestions.
🔍 Problem Summary

The root of the issue appears to be unconditional invocation of the address mapper for all requests, regardless of their origin:

// generic_DRAM_system.cpp
bool send(Request req) override {
  m_addr_mapper->apply(req);  // Always called
  int channel_id = req.addr_vec[0];
  ...
}

However:

    LoadStoreTrace constructs requests with valid addr, but empty addr_vec.

    ReadWriteTrace constructs requests with valid addr_vec, but addr = -1.

So when apply() is invoked on a ReadWriteTrace request, the address mapper slices from an invalid value -1, causing all entries in addr_vec to become max values — producing invalid DRAM addresses and memory mappings.
🧪 Reproduction Example

Using this ReadWriteTrace:

R 0,1,2,100,50
W 0,1,3,200,75

If you use any address mapper (e.g., ChRaBaRoCo), the manually crafted address vector will be overwritten by slicing addr = -1, which leads to incorrect and duplicated access patterns in simulation.
✅ Suggestions

Here are a few possible solutions:
✅ Option 1: Guard inside apply()

void apply(Request& req) override {
  if (req.addr == -1) {
    return;  // addr_vec is already valid
  }
  // mapping logic for physical addr
}

✅ Option 2: Add a flag to Request

struct Request {
  bool needs_mapping = true;
  ...
};

Then the mapper can skip apply() if !needs_mapping.
✅ Option 3: Handle in send() logic

bool send(Request req) override {
  if (req.addr != -1) {
    m_addr_mapper->apply(req);
  }
  ...
}

🙋 Question for You

Have you already worked around or resolved this issue in your setup?

If not, I’m happy to open a PR based on one of the suggestions above — but I’d love to hear your thoughts first to avoid duplication.

Thanks again for reporting this — very helpful!

YIAI-02, 2025-09-18:
Hello, I would also like to use ReadWriteTrace as the frontend implementation. However, when I execute the compiled exe file, I get the following error:
terminating due to uncaught exception of type Ramulator::ConfigurationError: 
Param "path" for implementation "ReadWriteTrace" is required but not given.
May I ask whether I need to make any configuration before build in order to use it?