CMU-SAFARI/ramulator2 #71: Incorrect Density for HBM2_8Gb Preset in Memory Configuration

Ozen7, 2024-11-16:
https://github.com/CMU-SAFARI/ramulator2/blob/e62c84a6f0e06566ba6e182d308434b4532068a5/src/dram/impl/HBM2.cpp#L14

Hello, 

The HBM2_8Gb preset defines a density of 6<<10 (6144 Mb or 6 Gb) per channel. However, the preset name (HBM2_8Gb) suggests the density should be 8<<10 (8192 Mb or 8 Gb) per channel. This mismatch leads to errors when using this preset in configurations expecting 8 Gb density.

### Steps to Reproduce
1) Configure Ramulator with the following setup:
```
DRAM:
  impl: HBM2
  org:
    preset: HBM2_8Gb
    channel: 8
    rank: 1
  timing:
    preset: HBM2_2000
```
2) Run a simulation with this configuration.
3) Observe the error indicating a mismatch between the calculated density and the preset density:
`An error occurred: Calculated HBM2 channel density 8192 Mb does not equal the provided density 6144 Mb!`

### Expected Behavior
The HBM2_8Gb preset should define a density of 8<<10 (8192 Mb) per channel, matching the preset name.

Thank you!

Ozen7, 2024-11-16:
Also, out of pure curiosity, can HBM be used with a closed-row policy? I realize it probably doesn't make much sense from a real-world use perspective, but I was wondering if it was at all possible.

For reference, the configuration errors out when you try to implement this since the closed-row policy requires a rank from the memory implementation (and HBM of course doesn't have one).

https://github.com/CMU-SAFARI/ramulator2/blob/e62c84a6f0e06566ba6e182d308434b4532068a5/src/dram_controller/impl/rowpolicy/basic_rowpolicies.cpp#L56

Mi5sssss, 2025-01-06:
yeah, the HBM3 configuration also got some problems here:

      //   name     density   DQ    Ch Pch  Bg Ba   Ro     Co
      {"HBM3_2Gb",   {2<<10,  128,  {1, 2,  4,  4, 1<<13, 1<<6}}},
      {"HBM3_4Gb",   {4<<10,  128,  {1, 2,  4,  4, 1<<14, 1<<6}}},
      {"HBM3_8Gb",   {6<<10,  128,  {1, 2,  4,  4, 1<<15, 1<<6}}},