CMU-SAFARI/ramulator2 #76: cmake issue of getting started

kijukim1, 2025-01-26:
Thank you for sharing the good simulator materials.
I have an issue that executing "cmake ..".
Is this related to yaml-cpp?
[image]

RichardLuo79, 2025-01-26:
Hi,

What is your cmake version?

kijukim1, 2025-01-26:
[image]
Thank you very much for quick response. It says that cmake version 3.31.20241015-gaeaeb83.
Is this compatible to run ramulator2?

RichardLuo79, 2025-01-26:
Hmm, the version should be new enough. I will investigate this issue tomorrow.

kijukim1, 2025-01-26:
That'll be great, thank you.
I can also try with previous cmake versions.

kijukim1, 2025-01-27:
[image]
I tried this with a previous cmake version but it shows same symptom.
However, I figured out that the issue was caused my execution without an installation of git.
So, I installed git with sudo apt install git.
And then "cmake .." works fine.

Thank you for looking into this, this thread can be closed.

closed 2025-01-27 by kijukim1