CMU-SAFARI/ramulator2 #77: Possibilities of running a customized workload on ramulator2

kijukim1, 2025-01-29:
Hello just a quick question.
Can I run a simple workload like vector add, matrix multiplication, etc on ramulator2?
My goal is evaluating matrices such as latency (s) and energy (J) to run a customized algorithm/workload on DRAM. I am planning yo fix simple vector add to make my workload.

RichardLuo79, 2025-02-02:
Hi,

I see three ways to do this:

1. Attach Ramulator2 to a detailed processor and system simulator (e.g., gem5).
2. Use the pin-based trace collector of Ramulator2 to collect the memory traces from your application, and use Ramulator2's SimpleO3 processor model frontend.
3. If your workload has static and regular memory access pattern, it might be feasible to synthetically generate the memory traces for Ramulator2 based on an analysis of the workload.

kijukim1, 2025-02-02:
Awesome, thank you very much for the great tips!

closed 2025-02-02 by kijukim1