CMU-SAFARI/ramulator2 #80: Memory Traces in Ramulator2

RavanN700, 2025-02-23:
Hello,

I want to run Ramulator2 with memory traces. I have looked at Ramulator (version 1) README, it supports memory traces:

```
Memory Trace Driven: Ramulator directly reads memory traces from a file, and simulates only the DRAM subsystem. Each line in the trace file represents a memory request, with the hexadecimal address followed by 'R' or 'W' for read or write.
0x12345680 R
0x4cbd56c0 W
```

I want to run a similar trace in Ramulator2, but it gives the following error:

```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoll
Aborted (core dumped)
```

Does Ramulator2 support memory traces? 

Thanks.

RichardLuo79, 2025-02-23:
Hi,

There are memory trace frontends under src/frontend/impl/memory_trace

RavanN700, 2025-02-23:
Hi,

Thanks for your response. As I looked at those trace frontends, it seems ramulator does get read/write (or load/store) memory traces. Do you know how I should write .yaml input file so that it gets the memory trace correctly?

My .yaml configuration file is similar to this one:

Frontend:
  impl: BHO3
  clock_ratio: 8
  num_expected_insts: 500000
  llc_capacity_per_core: 2MB
  llc_num_mshr_per_core: 16
  inst_window_depth: 128
  traces: 
    - example_inst.trace
  no_wait_traces:
    - example_rh_physaddr.trace

Thanks.

closed 2025-02-23 by RavanN700

TurnOffNOD, 2025-03-14:
> Hi,
> 
> There are memory trace frontends under src/frontend/impl/memory_trace

And how to use it? How can we generate dram traces with our custom executable?