CMU-SAFARI/ramulator2 #83: Why tCK_ps in LPDDR5 is always recalculated and is different from the default value in timing_presets

EricHnbc, 2025-03-21:
In src/dram/impl/LPDDR5.cpp, LPDDR5_6400 timing parameters are defined in line 21 and the tCK_ps is 1250 ps.

      //   name         rate   nBL  nCL  nRCD  nRPab  nRPpb   nRAS  nRC   nWR  nRTP nCWL nCCD nRRD nWTRS nWTRL nFAW  nPPD  nRFCab nRFCpb nREFI nPBR2PBR nPBR2ACT nCS,  tCK_ps
      {"LPDDR5_6400",  {6400,  4,   20,   15,    17,   15,     34,   30,   28,   4,  11,   4,   4,   5,    10,   16,  2,   -1,      -1,   -1,   -1,        -1,    2,   1250}},

But in line 258-259, tCK_ps is recalculated and it is changed to 312.5 ps for LPDDR5_6400.

      int tCK_ps = 1E6 / (m_timing_vals("rate") / 2);
      m_timing_vals("tCK_ps") = tCK_ps;

Is there any reason to recalculate the tCK_ps and why it is different from the default value 1250 ps? Should these two lines be moved into previous braces to recalculate it only when "rate" is re-configured?

When calculating tCK_ps, do you consider the 4:1 ratio between WCK and CK? Based on the JEDEC LPDDR5/5X spec, LPDDR5_6400 only support 4:1 ratio and CK clock frequency is 800 MHz, so 1250 ps should be the correct value for tCK_ps.

kiriakaras, 2025-03-25:
Hi, Im just another user, I believe it is recalculated to match the device clock frequency of 3200Mhz. Take my words with a pinch of salt, but the frequency for the DMC is indeed 800Mhz and this translates to a clock cycle of 1250 ps, but the internal (device) clock used in the LPDDR5_6400  is 3200Mhz or 312.5 ps (6400 stands for Double Data Rate).

kiriakaras, 2025-03-25:
I've also printed most of the specified timings, and their values are also multiplied to match the internal clock rather than the DMC clock.

RichardLuo79, 2026-05-14:
Hi, thanks for checking! It is indeed a bug in v2.0 where the tCK_ps calculation did not handle WCK:CK = 4:1 case.

We recommend switching to v2.1 (https://github.com/CMU-SAFARI/ramulator2/tree/v2.1) where this issue no longer exists. v2.1 also comes with better LPDDR5 controller support.

closed 2026-05-14 by RichardLuo79