CMU-SAFARI/ramulator2 #86: Installation on arm macbook

BryceYuZh, 2025-04-06:
Hi,
When I do
cmake .. -DCMAKE_CXX_COMPILER=clang-15 -DCMAKE_CXX_STANDARD=20 -DCMAKE_CXX_FLAGS="-std=c++20 -stdlib=libc++"
I got the error during make and could not resolve:

...

 Ramulator::Factory::create_implementation(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, YAML::Node const&, Ramulator::Implementation*) in factory.cpp.o
      Ramulator::Factory::create_frontend(YAML::Node const&) in factory.cpp.o
      Ramulator::TopLevel<Ramulator::IFrontEnd>::gather_components() in factory.cpp.o
      Ramulator::Factory::create_memory_system(YAML::Node const&) in factory.cpp.o
      ...
  "___dynamic_cast", referenced from:
      Ramulator::Factory::create_frontend(YAML::Node const&) in factory.cpp.o
      Ramulator::TopLevel<Ramulator::IFrontEnd>::gather_components() in factory.cpp.o
      Ramulator::Factory::create_memory_system(YAML::Node const&) in factory.cpp.o
      Ramulator::TopLevel<Ramulator::IMemorySystem>::gather_components() in factory.cpp.o
      Ramulator::TestIfce* Ramulator::Implementation::create_child<Ramulator::TestIfce>(YAML::Node const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) in test_impl.cpp.o
      Ramulator::TestImpl2* Ramulator::Implementation::create_child_impl<Ramulator::TestIfce, Ramulator::TestImpl2>(YAML::Node const&) in test_impl.cpp.o
      Ramulator::ITranslation* Ramulator::Implementation::create_child<Ramulator::ITranslation>(YAML::Node const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) in simpleO3.cpp.o
      ...
  "___gxx_personality_v0", referenced from:
      std::__1::__throw_length_error[abi:v15007](char const*) in factory.cpp.o
      std::__1::__throw_out_of_range[abi:v15007](char const*) in factory.cpp.o
      fmt::v9::detail::throw_format_error(char const*) in factory.cpp.o
      Dwarf Exception Unwind Info (__eh_frame) in factory.cpp.o
      std::__1::__throw_length_error[abi:v15007](char const*) in logging.cpp.o
      fmt::v9::detail::throw_format_error(char const*) in logging.cpp.o
      Dwarf Exception Unwind Info (__eh_frame) in logging.cpp.o
      ...

ld: symbol(s) not found for architecture arm64
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../libramulator.dylib] Error 1
make[1]: *** [CMakeFiles/ramulator.dir/all] Error 2
make: *** [all] Error 2

Could you please help.

codingMMax, 2025-04-28:
I encountered the same error.

codingMMax, 2025-04-28:
I successfully built the repo in my M2Pro Macbook Pro 16 on macOS Sequoia 15.3.1, theck out the #44 .
# install and activate upstream Clang 15
brew install llvm@15
echo 'export PATH="/opt/homebrew/opt/llvm@15/bin:$PATH"' >> ~/.zshrc
export CC=/opt/homebrew/opt/llvm@15/bin/clang
export CXX=/opt/homebrew/opt/llvm@15/bin/clang++
hash -r     # refresh your shell’s command cache

inside ramulator2/build, run:
cmake  -DCMAKE_C_COMPILER=$CC \
       -DCMAKE_CXX_COMPILER=$CXX \
       -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_OSX_ARCHITECTURES=arm64 \
       ..

and build as normal.

RichardLuo79, 2025-05-28:
Hi,

We recommend using the Docker container (Dockerhub: `richardluo831/ramulator2`) to avoid issues like these. This image is multi-platform, and we have tested on a Apple Silicon MacBook Pro.

Best Regards,
Haocong

closed 2025-05-28 by RichardLuo79